waterloo-rocketry/canhw #22: RF footprint needs to be improved

JaredWatson, 2020-05-10:
Currently the RF footprint used on the GPS board and the Live Telemetry board had extremely small pads that make soldering it a pain in the butt. In the future the amount of soldering space should probably be expanded (but not enough to shrink the space of the RF trace, it's already smaller than 120 mils and we don't want to lose that sweet sweet RF integrity).

QuantumManiac, 2020-05-11:
I did a bit of testing, and it seems like it's pretty simple to have different sized pads on different layers.

In the _Pad Properties_ window, there's a Copper setting which allows you to change the layers that the copper and other layers are on for the pad:
[image]

My solution is to overlap two pads at each through-hole, with one pad having F.Cu and F.Mask and the other B.Cu and B.Mask 

Here's a view of a modified SMA connector in the footprint editor modified using the above method:
[image]
Notice that while the front pads are 2.2 mm, the back ones are much larger, at 3 mm 

3D Views - front and back comparison:
[image]

If I got some measurements for what the pads should be on each side, I would be happy to modify the footprint.

JaredWatson, 2020-05-12:
Looks good, I'll leave a more formal review on the actual PR. This is definitely the footprint people should be using for SMA stuff in the future.

closed 2020-05-21 by zx100x100